dmitrystu/sboot_stm32 #5: Boot pin inversion

sn00pster, 2019-08-06:
Boot pin is configured for LOW to enter bootloader, I propose a change to allow the logic to be inverted from the config file.

stm32l4xx.S (as an example)

```
.L_scan_bootstrap:
    ldr     r2, [r1, #GPIO_IDR] 
 #if (DFU_BOOTSTRAP_GPIO_ACTIVE_LEVEL == _HIGH)
    eor     r2, #(1<<(DFU_BOOTSTRAP_PIN))
#endif
    lsrs    r2, #(DFU_BOOTSTRAP_PIN + 1)    //Pin -> CF
```

and in config.h

```
#define _HIGH 1
#define _LOW 0

#define DFU_BOOTSTRAP_GPIO_ACTIVE_LEVEL   _HIGH
```

dmitrystu, 2019-08-06:
Also pin pulldown must be configured.

sn00pster, 2019-08-06:
> Also pin pulldown must be configured.

probably should be an option to configure pull up, down or none.

in our case on our hardware the pin has an external pull down which is weak as we are current concious and every microamp counts.

i have forked this, i have some other changes i am going to implement so will send a pull request with those when i am done.

dmitrystu, 2019-08-06:
Pullup on the bootstrap pin is enabled for the very short time after reset.
FYI. You don't need to invert pin readout in the R2. Just replace cbz by cbnz in the line 173
https://github.com/dmitrystu/sboot_stm32/blob/master/mcu/stm32l4xx.S#L173

sn00pster, 2019-08-09:
I've created a pull request with this functionality:

https://github.com/dmitrystu/sboot_stm32/pull/7

dmitrystu, 2019-08-10:
It was a my mistake to use cbnz for the high level. It will break doublereset and softreset activation. I have made correct commit. Tested with F103 and L433.

sn00pster, 2019-08-10:
Ok, cool, I will use this and will pull in the changes and disregard mine for the pullup/down configuration.  Many thanks for this

sorokolokarim, 2019-08-12:
Thank you for your answer. But after to have search and search, i arrive now to launch make command. But i have many errors when i run specicy CMD you have given : ACER i5@SORO_KOLO MINGW64 ~/Desktop/sboot_stm32-master
$ make stm32l052x8
C:/Program Files/Git/mingw64/bin/make fwclean bootloader FWCPU='-mcpu=cortex-m0plus' \
                           FWSTARTUP='mcu/stm32l0xx.S' \
                           FWDEFS='STM32L0 STM32L052xx USBD_ASM_DRIVER' \
                           LDPARAMS='ROMLEN=64K RAMLEN=8K'
make[1]: Entering directory 'C:/Users/ACER i5/Desktop/sboot_stm32-master'
compiling src/arc4.c
compiling src/chacha.c
compiling src/gost.c
compiling src/raiden.c
compiling src/rc5.c
compiling src/speck.c
compiling src/xtea.c
compiling src/blowfish.c
compiling src/rtea.c
assembling mcu/stm32l0xx.S
compiling src/descriptors.c
src/descriptors.c:19:19: fatal error: stm32.h: No such file or directory
 #include "stm32.h"
                   ^
compilation terminated.
make[1]: *** [Makefile:110: build/objfw/descriptors.o] Error 1
make[1]: Leaving directory 'C:/Users/ACER i5/Desktop/sboot_stm32-master'
make: *** [Makefile:263: stm32l052x8] Error 2

dmitrystu, 2019-08-12:
It's easy. You needs
1. [CMSISV4](https://github.com/ARM-software/CMSIS) or [CMSISV5](https://github.com/ARM-software/CMSIS_5) from ARM
2. Hardware definition includes for your MCU. Can be obtained from ST as a part of the Stcube. It must be placed into corresponded folder of CMSIS
3. [stm32.h](https://github.com/dmitrystu/stm32h)
4. Set environmental variable CMSIS to point to CMSIS root folder.
 `````
Somepath/Cmsis root   <--- CMSIS root. Point CMSIS envar here.
  +-CMSIS              <--- generic CMSIS from ARM
  +-Device
    +-ST               <--- ST Vendor folder
      +-STM32F0xx
      +-STM32L0xx
      +- stm32.h
   
`````

sorokolokarim, 2019-08-12:
Hi, I still wanted to thank you for the great job you did and also for the attention you pay to my requests. So I must say that it's probably easy for you all that, but me now I risk exploding a thermometer if it approaches near my head (so I'm struggling to get out). It's very new to me it's line of code, until then I was abitute to run programs and it's all or more to write some lines with arduino and that's it! Please help me, I spend sleepless nights now. To begin: I downloaded CMSISV5 from ARM, I work with the STM32F103T8, and I would like to install a secure bootloader (not to mention that just a few days ago I started in the stm32 universe) and to tell the truth, I can not exploit the information that you have just given me. I am very self taught, and I am sure that by realizing some examples for me, then I will be able to gradually get out of it: - / thank you sincerely

sorokolokarim, 2019-08-14:
Hello ! How are you ! Thank you for your bigest help you have given to me. I am on the way to build a bootloader from your project, but my problem now is : "4 Set environmental variable CMSIS to point to CMSIS root folder". Thank you for your help 👍

dmitrystu, 2019-08-17:
Hi. Sorry for the late reply.
There is some options to do this:
1. Via system environmental variables. It's a preferred way for me.
2. pass CMSIS to mingw32-make. ```mingw32-make CMSIS='C:/path/to/the/cmsis/root'```
3. modify line 5 in the project makefile. ```CMSIS      ?= $(abspath ../CMSIS)  -> CMSIS ?= 'C:/path/to/the/cmsis/root'```
PS. Not sure about using quotes under Win.

sorokolokarim, 2019-08-18:
Hello !!! Thank you for your kindness ! I have done all you told to me 
[image]

dmitrystu, 2019-08-18:
Did you update USB submodule?
```git submodule update --init```

sorokolokarim, 2019-08-18:
Sorry, but i don't know something about it, i starting in programmation ! So, how to do that ? :-/

closed 2019-09-04 by dmitrystu