dmitrystu/sboot_stm32 #11: Can't upload

Mozzytec, 2019-10-03:
Hello, 
After lot's of pain, I succeded to make the bin file for the bootloader. 
I flash the memory and the STM32F103C8 is recognized as a STM DFU. To test I don't use the encrytpion yet.
I remap the user programm to 0x08001000, compile with latest version of arduino ide. Download via DFU is not possible via Arduino. simple blink example led on PB13 and PC13
If I flash via DFU-Util on C:\dfu-util with admin right, the blink programm does not start at all. Any idea? 
COuld you provide a blink.bin to test the download with your file at least to test if the bootload operates correctly. I supect an error in my link file, but where ???
What should be done or use to compile all correctly?

dmitrystu, 2019-10-04:
[Here](https://ufile.io/s7eqnonc) is the binary for F103 I used to test. In fact, this is a demo from the [USB library](https://github.com/dmitrystu/libusb_stm32) linked with 0x08001000 base address(linker script also attached). This is not a blinking led but USB CDC+HID example.

dmitrystu, 2019-10-04:
Also check config.h. To disable encryptionn set 
```
#define DFU_CIPHER          _DISABLE
```

Mozzytec, 2019-10-04:
Hello
CIpher was disable for sure.
I do the test immediatly and let you knwow (ps Thanks for PM)

Mozzytec, 2019-10-04:
I have renewed all my usb Device base. I have plugged device is recognized as a STM device in DFU mode, but there is also an unknown device on top.

dmitrystu, 2019-10-05:
Perhaps I need to check and update DFU descriptors to get more compatibility with Windows.

dmitrystu, 2019-10-06:
Tested the latest master on Win10Home V1703 build 15063.1387. No problems such as YB and unrecognized devices found. Bluepill (STM32F103C8T6) has been detected as 'STM device in DFU mode'  (STTub30.sys)

closed 2019-12-04 by dmitrystu