dmitrystu/sboot_stm32 #17: Platform init

fizzyade, 2020-05-19:
Ideally the boot loader should provide a platform_init() type function (by default it does nothing and is weak) which is executed before any of the boot loader verification functions are run, ideally it should be called as early as possible.

The idea behind this is that it would allow the system clock to be configured to the fastest available for the platform, this would have a dramatic effect things like PKI signature verification.

dmitrystu, 2020-05-19:
I accept this point, but in other side some sort of the platform_deinit() must be added to revert back settings if bootloader start is not required. It will enlarge code.
But i have an idea how to to this other way.

dmitrystu, 2020-05-20:
Could you check [this](https://github.com/dmitrystu/sboot_stm32/tree/dev/early_clock_setup) experimental branch? Done for F4.

fizzyade, 2020-05-21:
I'll have to look ASAP Dmitry.  I've only just returned to the project that is going to be using this and it's taken me a while to remember where I was and what I was up to! 

I'm just adding support into our application firmware for the DFU runtime descriptors and control endpoint commands that are needed, then I will test this, although this particular board is using a L4 processor.

dmitrystu, 2020-05-21:
BTW L4 is done.

dmitrystu, 2020-08-19:
This feature was implemented. I will close this issue.

closed 2020-08-19 by dmitrystu