dmitrystu/sboot_stm32 #19: cannot open linker script file build/objfw/script.

withgallantry, 2020-07-01:
Hi, firstly, thank you for creating this, it's just what I've been looking for.
I'm having trouble compiling it. I am running on macOS Catalina v10.15.2.

I'm having trouble building the linker.

```
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f ldscript.mk ROMLEN=64K RAMLEN=20K OUTFILE=build/objfw/script.ld
Building linker script
building bootloader
/usr/local/Cellar/arm-none-eabi-gcc/9-2019-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: cannot open linker script file build/objfw/script.ld: No such file or directory
```

It seems script.ld is not being generated.
Any insight you can give would be greatly appreciated. Thanks.

dmitrystu, 2020-07-01:
What version of the make you are using? ```make -v```

dmitrystu, 2020-07-01:
It might be a problem with the syntax of the ```$file``` function. Could you change https://github.com/dmitrystu/sboot_stm32/blob/master/ldscript.mk#L76 to ```$(file >$@,$(LDSCRIPT))``` and try again.

withgallantry, 2020-07-02:
Thanks @dmitrystu It was my make version. I was previously running version 3.81 but I upgraded to 4.3 and the issue was resolved. Thanks for your time with this.

withgallantry, 2020-07-02:
Got it all compiled and flashed, except it doesn't seem to be in DFU mode, using a bluepill, so compiling for STM32F103C8. Doesn't show up in dfu-util list.

dmitrystu, 2020-07-02:
If you using default configuration (without checksum verification) you must activate bootloader manually to get DFU interface. Doubletap RESET button or tie A1 to GND then reset.

closed 2020-07-09 by dmitrystu