dmitrystu/sboot_stm32 #21: Build fails with linker errors

Polsaker, 2020-07-23:
```
make bootloader FWCPU='-mcpu=cortex-m3' \
                   FWSTARTUP='mcu/stm32f105.S' \
                   FWDEFS='STM32F1 STM32F107xC HSE_25MHZ USBD_VBUS_DETECT' \
                   LDPARAMS='ROMLEN=128K RAMLEN=20K'
make[1]: Entering directory '/home/polsaker/build/sboot_stm32'
compiling src/arc4.c
compiling src/chacha.c
compiling src/gost.c
compiling src/raiden.c
compiling src/rc5.c
compiling src/speck.c
compiling src/xtea.c
compiling src/xtea1.c
compiling src/blowfish.c
compiling src/rtea.c
compiling src/rc6.c
compiling src/rijndael.c
compiling src/magma.c
compiling src/checksum.c
compiling src/crypto.c
assembling mcu/stm32f105.S
compiling src/descriptors.c
In file included from usb/inc/usb.h:22,
                 from src/descriptors.c:21:
usb/inc/usbd_core.h:38:1: warning: "/*" within comment [-Wcomment]
   38 | /** @} */
      |  
compiling src/bootloader.c
In file included from usb/inc/usb.h:22,
                 from src/bootloader.c:21:
usb/inc/usbd_core.h:38:1: warning: "/*" within comment [-Wcomment]
   38 | /** @} */
      |  
assembling src/rc5a.S
assembling src/chacha_a.S
assembling src/rc6a.S
building module usb
make[2]: Entering directory '/home/polsaker/build/sboot_stm32/usb'
make /home/polsaker/build/sboot_stm32/build/objfw/libusb.a
make[3]: Entering directory '/home/polsaker/build/sboot_stm32/usb'
compiling src/usbd_core.c
compiling src/usbd_stm32f103_devfs.c
compiling src/usbd_stm32f105_otgfs.c
src/usbd_stm32f105_otgfs.c: In function 'ep_read':
src/usbd_stm32f105_otgfs.c:335:13: warning: 'packed' attribute ignored for type 'uint32_t *' {aka 'long unsigned int *'} [-Wattributes]
  335 |             *(__attribute__((packed))uint32_t*)buf = _t;
      |             ^
src/usbd_stm32f105_otgfs.c: In function 'ep_write':
src/usbd_stm32f105_otgfs.c:364:9: warning: 'packed' attribute ignored for type 'uint32_t *' {aka 'long unsigned int *'} [-Wattributes]
  364 |         *_fifo = *(__attribute__((packed)) uint32_t*)buf;
      |         ^
compiling src/usbd_stm32f429_otgfs.c
compiling src/usbd_stm32f429_otghs.c
compiling src/usbd_stm32f446_otgfs.c
compiling src/usbd_stm32l052_devfs.c
compiling src/usbd_stm32l100_devfs.c
compiling src/usbd_stm32l433_devfs.c
compiling src/usbd_stm32l476_otgfs.c
assembling src/usbd_stm32f103_devfs_asm.S
assembling src/usbd_stm32l052_devfs_asm.S
assembling src/usbd_stm32l100_devfs_asm.S
a - obj/usbd_core.o
a - obj/usbd_stm32f103_devfs.o
a - obj/usbd_stm32f105_otgfs.o
a - obj/usbd_stm32f429_otgfs.o
a - obj/usbd_stm32f429_otghs.o
a - obj/usbd_stm32f446_otgfs.o
a - obj/usbd_stm32l052_devfs.o
a - obj/usbd_stm32l100_devfs.o
a - obj/usbd_stm32l433_devfs.o
a - obj/usbd_stm32l476_otgfs.o
a - obj/usbd_stm32f103_devfs_asm.o
a - obj/usbd_stm32l052_devfs_asm.o
a - obj/usbd_stm32l100_devfs_asm.o
make[3]: Leaving directory '/home/polsaker/build/sboot_stm32/usb'
make[2]: Leaving directory '/home/polsaker/build/sboot_stm32/usb'
make -f ldscript.mk ROMLEN=128K RAMLEN=20K OUTFILE=build/objfw/script.ld
make[2]: Entering directory '/home/polsaker/build/sboot_stm32'
Building linker script
make[2]: Leaving directory '/home/polsaker/build/sboot_stm32'
building bootloader
/usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: build/objfw/bootloader.o (symbol from plugin): in function `main':
(.text+0x0): multiple definition of `aes_blksize'; build/objfw/crypto.o (symbol from plugin):(.text+0x0): first defined here
/usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: build/objfw/bootloader.o (symbol from plugin): in function `main':
(.text+0x0): multiple definition of `aes_name'; build/objfw/crypto.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:120: build/firmware.elf] Error 1
make[1]: Leaving directory '/home/polsaker/build/sboot_stm32'
make: *** [Makefile:469: stm32f107xb] Error 2

```

happened in the stm32f107xb target

dmitrystu, 2020-07-24:
See no problems with gcc version 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] (GNU Tools for Arm Embedded Processors 9-2019-q4-major). 

Try to add an explicit ```extern``` keyword to  the ```aes_blksize``` and ```aes_name``` declaration in the inc/crypto.h

Polsaker, 2020-07-24:
That worked, thanks!

I also had to do it for `checksum_name` and `checksum_length` in inc/checksum.h

closed 2020-07-24 by Polsaker