dmitrystu/sboot_stm32 #25: Error "sorry - this program has been built without plugin support"

curiousmindsdev, 2020-10-26:
Hi Dmitry,

I am trying to compile this for STM32F070CB. “make stm32f070xb” command gives me  “/bin/sh: del: command not found        make: *** [fwclean] Error 127” error.
If I type “make bootloader stm32f070xb”, it compiles everything until it starts to build usb module.
Error log is below:
“ building module usb
make[1]: Entering directory `/c/Users/cdev/Desktop/STM32_BL/usb'
/bin/sh: del: command not found
make[1]: *** [clean] Error 127
make[1]: Leaving directory `/c/Users/cdev/Desktop/STM32_BL/usb'
make: *** [build/objfw/libusb.a] Error 2 ”

I replaced 
--------------------------
ifeq ($(OS),Windows_NT)
	RM = del /Q
	fixpath = $(strip $(subst /,\, $1))
else
	RM = rm -f
	fixpath = $(strip $1)
endif
-----------------------
with
------------------------
	RM = rm -f
	fixpath = $(strip $1)
------------------------
in both usb and main makefile.

Now I get this error.

C:\Users\cdev\Desktop\STM32_BL>make stm32f070xb
make bootloader FWCPU='-mcpu=cortex-m0' \
                           FWSTARTUP='mcu/stm32f0xx.S' \
                           FWDEFS='STM32F0 STM32F070xB USBD_ASM_DRIVER' \
                           LDPARAMS='ROMLEN=128K RAMLEN=16K'
make[1]: Entering directory `/c/Users/cdev/Desktop/STM32_BL'
compiling src/arc4.c
compiling src/chacha.c
compiling src/gost.c
compiling src/raiden.c
compiling src/rc5.c
compiling src/speck.c
compiling src/xtea.c
compiling src/xtea1.c
compiling src/blowfish.c
compiling src/rtea.c
compiling src/rc6.c
compiling src/rijndael.c
compiling src/magma.c
compiling src/checksum.c
compiling src/crypto.c
assembling mcu/stm32f0xx.S
compiling src/descriptors.c
In file included from usb/inc/usb.h:22:0,
                 from src/descriptors.c:21:
usb/inc/usbd_core.h:38:1: warning: "/*" within comment [-Wcomment]
compiling src/bootloader.c
In file included from usb/inc/usb.h:22:0,
                 from src/bootloader.c:21:
usb/inc/usbd_core.h:38:1: warning: "/*" within comment [-Wcomment]
assembling src/rc5a.S
assembling src/chacha_a.S
assembling src/rc6a.S
building module usb
make[2]: Entering directory `/c/Users/cdev/Desktop/STM32_BL/usb'
make /c/Users/cdev/Desktop/STM32_BL/build/objfw/libusb.a
make[3]: Entering directory `/c/Users/cdev/Desktop/STM32_BL/usb'
compiling src/usbd_core.c
compiling src/usbd_stm32f103_devfs.c
compiling src/usbd_stm32f105_otgfs.c
compiling src/usbd_stm32f429_otgfs.c
compiling src/usbd_stm32f429_otghs.c
compiling src/usbd_stm32f446_otgfs.c
compiling src/usbd_stm32f446_otghs.c
compiling src/usbd_stm32l052_devfs.c
src/usbd_stm32l052_devfs.c:24:13: note: #pragma message: PMA memory size is not defined. Use 1k by default
compiling src/usbd_stm32l100_devfs.c
compiling src/usbd_stm32l433_devfs.c
compiling src/usbd_stm32l476_otgfs.c
assembling src/usbd_stm32f103_devfs_asm.S
assembling src/usbd_stm32l052_devfs_asm.S
assembling src/usbd_stm32l100_devfs_asm.S
sorry - this program has been built without plugin support
make[3]: *** [/c/Users/cdev/Desktop/STM32_BL/build/objfw/libusb.a] Error 24
make[3]: Leaving directory `/c/Users/cdev/Desktop/STM32_BL/usb'
make[2]: *** [module] Error 2
make[2]: Leaving directory `/c/Users/cdev/Desktop/STM32_BL/usb'
make[1]: *** [build/objfw/libusb.a] Error 2
make[1]: Leaving directory `/c/Users/cdev/Desktop/STM32_BL'
make: *** [stm32f070xb] Error 2

Can you help me with this please? Also I tried to import this to a Atollic TrueSTUDIO or STM32cubeIDE. Didn’t managed to do it properly. I am new to this. Recently transitioned from Arduino IDE. Can you guide me to import this to an IDE? Thanks for your time.

dmitrystu, 2020-10-26:
Looks like you are under Windows with WSL2. I faced a similar problem when I tried to configure CI (Giithub Actions). Check [this](https://github.com/dmitrystu/sboot_stm32/blob/master/.github/workflows/test-build-win.yml#L27) script. It may be helpful

curiousmindsdev, 2020-10-27:
It didn't help. I'm really not sure why I'm getting this error. Do you have any compiled hex file for stm32f070cb? Will I get this error if run the makefile in linux?

dmitrystu, 2020-10-27:
What's version of the make you are using ? I just built bootloader under Windows 10 using both mingw32-make and GNU make. See no any problems.
```
PS C:\proj\sboot_stm32> arm-none-eabi-gcc.exe -v
Using built-in specs.
COLLECT_GCC=C:\Program Files (x86)\GNU Arm Embedded Toolchain\9 2020-q2-update\bin\arm-none-eabi-gcc.exe
COLLECT_LTO_WRAPPER=c:/program\ files\ (x86)/gnu\ arm\ embedded\ toolchain/9\ 2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/lto-wrapper.exe
Target: arm-none-eabi
Configured with: /mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/src/gcc/configure --build=x86_64-linux-gnu --host=i686-w64-mingw32 --target=arm-none-eabi --prefix=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-mingw --libexecdir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-mingw/lib --infodir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-mingw/share/doc/gcc-arm-none-eabi/info --mandir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-mingw/share/doc/gcc-arm-none-eabi/man --htmldir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-mingw/share/doc/gcc-arm-none-eabi/html --pdfdir=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-mingw/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-mingw-wildcard --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-headers=yes --with-newlib --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/install-mingw/arm-none-eabi --with-libiconv-prefix=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-mingw/host-libs/usr --with-gmp=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-mingw/host-libs/usr --with-mpfr=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-mingw/host-libs/usr --with-mpc=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-mingw/host-libs/usr --with-isl=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-mingw/host-libs/usr --with-libelf=/mnt/workspace/workspace/GCC-9-pipeline/jenkins-GCC-9-pipeline-200_20200521_1590053374/build-mingw/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='GNU Arm Embedded Toolchain 9-2020-q2-update' --with-multilib-list=rmprofile,aprofile
Thread model: single
gcc version 9.3.1 20200408 (release) (GNU Arm Embedded Toolchain 9-2020-q2-update)

PS C:\proj\sboot_stm32> mingw32-make -v
GNU Make 3.82.90
Built for i686-pc-mingw32
Copyright (C) 1988-2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

PS C:\proj\sboot_stm32> make -v
GNU Make 4.3
Built for Windows32
Copyright (C) 1988-2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

PS C:\proj\sboot_stm32> make stm32f070xb
C:/ProgramData/chocolatey/lib/make/tools/install/bin/make.exe bootloader FWCPU='-mcpu=cortex-m0' \
                   FWSTARTUP='mcu/stm32f0xx.S' \
                   FWDEFS='STM32F0 STM32F070xB USBD_ASM_DRIVER' \
                   LDPARAMS='ROMLEN=128K RAMLEN=16K'
make[1]: Entering directory 'C:/proj/sboot_stm32'
compiling src/arc4.c
compiling src/chacha.c
compiling src/gost.c
compiling src/raiden.c
compiling src/rc5.c
compiling src/speck.c
compiling src/xtea.c
compiling src/xtea1.c
compiling src/blowfish.c
compiling src/rtea.c
compiling src/rc6.c
compiling src/rijndael.c
compiling src/magma.c
compiling src/checksum.c
compiling src/crypto.c
assembling mcu/stm32f0xx.S
compiling src/descriptors.c
In file included from usb/inc/usb.h:22,
                 from src/descriptors.c:21:
usb/inc/usbd_core.h:38:1: warning: "/*" within comment [-Wcomment]
   38 | /** @} */
      |
compiling src/bootloader.c
In file included from usb/inc/usb.h:22,
                 from src/bootloader.c:21:
usb/inc/usbd_core.h:38:1: warning: "/*" within comment [-Wcomment]
   38 | /** @} */
      |
assembling src/rc5a.S
assembling src/chacha_a.S
assembling src/rc6a.S
building module usb
make[2]: Entering directory 'C:/proj/sboot_stm32/usb'
Не удается найти C:\proj\sboot_stm32\usb\cdc_loop.*
C:/ProgramData/chocolatey/lib/make/tools/install/bin/make.exe C:/proj/sboot_stm32/build/objfw/libusb.a
make[3]: Entering directory 'C:/proj/sboot_stm32/usb'
compiling src/usbd_core.c
[... 19 lines cut ...]
a - obj/usbd_stm32f429_otgfs.o
a - obj/usbd_stm32f429_otghs.o
a - obj/usbd_stm32f446_otgfs.o
a - obj/usbd_stm32f446_otghs.o
a - obj/usbd_stm32l052_devfs.o
a - obj/usbd_stm32l100_devfs.o
a - obj/usbd_stm32l433_devfs.o
a - obj/usbd_stm32l476_otgfs.o
a - obj/usbd_stm32f103_devfs_asm.o
[... cut ...]
make[2]: Leaving directory 'C:/proj/sboot_stm32/usb'
C:/ProgramData/chocolatey/lib/make/tools/install/bin/make.exe -f ldscript.mk ROMLEN=128K RAMLEN=16K OUTFILE=build/objfw/script.ld
make[2]: Entering directory 'C:/proj/sboot_stm32'
Building linker script
make[2]: Leaving directory 'C:/proj/sboot_stm32'
building bootloader
   text    data     bss     dec     hex filename
   3700     124     364    4188    105c build/firmware.elf
creating build/firmware.hex
creating build/firmware.bin
make[1]: Leaving directory 'C:/proj/sboot_stm32'

```

curiousmindsdev, 2020-10-28:
My GNU make version was 3.8 . Updated it to 4.3 and it compiled it. I uploaded bootloader to STM32F070CB via ST-LINK using STM32Cube Programmer. My USB device shows up in both STM32Cube Programmer and Device manager. It appears in STM32Cube Programmer only when using ST's VID&PID, but when I try to upload my firmware to microcontroller I get "Error: Data read failed" error. Should I be able to upload using that way or should I develop my own firmware upload client for it?

Error log:
```
09:06:12 : USB speed : Full Speed (12MBit/s)
09:06:12 : Manuf. ID : CMD.
09:06:12 : Product ID : BioniX
09:06:12 : SN : 6EDB1566
09:06:12 : FW version : 0x0110
09:06:12 : IAP
09:06:12 : UPLOADING ...
09:06:12 : Size : 1024 Bytes
09:06:12 : Address : 0x8000000
09:06:12 : Read progress:
09:06:12 : Error: Data read failed
```

dmitrystu, 2020-10-28:
Yes. This bootloader implements a standard DFU interface, not a DFUSE from ST.
You can use dfu-util.

curiousmindsdev, 2020-10-30:
Tried dfu-util today. I got success with uploading a firmware with ST's built-in bootloader, but the sboot_stm32 bootloader failed to upload. When I list the available DFU devices, it seems that ST' BL gives me `"@Internal Flash  /0x08000000/064*0002Kg"` and the sboot_stm32 BL gives me only `"Internal Flash"`. I'm guessing that is only a name for the device. 

Log for firmware upload with ST's built-in bootloader:
```
C:\Users\cdev\Desktop\dfu-util-0.9-win64>dfu-util -l
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [0483:df11] ver=2200, devnum=6, cfg=1, intf=0, path="3-1.5", alt=1, name="@Option Bytes  /0x1FFFF800/01*016 e", serial="FFFFFFFEFFFF"
Found DFU: [0483:df11] ver=2200, devnum=6, cfg=1, intf=0, path="3-1.5", alt=0, name="@Internal Flash  /0x08000000/064*0002Kg", serial="FFFFFFFEFFFF"

C:\Users\cdev\Desktop\dfu-util-0.9-win64>dfu-util -d 0483:df11 -a "@Internal Flash  /0x08000000/064*0002Kg" --dfuse-address 0x08000000 -D demo2.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuDNLOAD-IDLE, status = 0
aborting previous incomplete transfer
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 3316
Download        [=========================] 100%         3316 bytes
Download done.
File downloaded successfully
```

Log for firmware upload with sboot-stm32:
```
C:\Users\cdev\Desktop\dfu-util-0.9-win64>dfu-util -l
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [0483:df11] ver=0100, devnum=3, cfg=1, intf=0, path="3-1.5", alt=0, name="Internal Flash", serial="6EDB1566"

C:\Users\cdev\Desktop\dfu-util-0.9-win64>dfu-util -d 0483:df11 -a "Internal Flash" --dfuse-address 0x08001000 -D demo-8001000.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 128
Could not read name, sscanf returned 0
Failed to parse memory layout

C:\Users\cdev\Desktop\dfu-util-0.9-win64>dfu-util -d 0483:df11 -a "Internal Flash" --dfuse-address 0x08002000 -D demo-8002000.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 128
Could not read name, sscanf returned 0
Failed to parse memory layout
```

dmitrystu, 2020-10-30:
As I said, this bootloader implements a standard DFU, not a DFUSE. This is a different protocol.
Do not use ```--dfuse-address``` nor ```-s``` options. You don't need to set up the address of the user firmware, it's already bond.

curiousmindsdev, 2020-10-30:
Thank you so much for your big help!!! I removed address options and it programmed successfully. I am quite new to this stuff but I learned a lot while researching for bootloader development. I'll try to add fuctionality to this bootloader so that I can update firmware of devcies connected to this main controller. I'll release it as open source if I can accomplish my goals. One last question. Is there a specific app start address? My bootloader size is 3348 Bytes, I tried uploading a demo LED toggle demo to addresses 0x08001000 and 0x08002000 and it didn't start. I'm guessing that I should add some extra lines of code to my firmware as well.

dmitrystu, 2020-10-30:
Check the build/firmware.map file for the ```__app_start``` label. You also can set an application start address manually using [config](https://github.com/dmitrystu/sboot_stm32/blob/master/CONFIG.md). Make sure it's aligned to the flash page.
BTW. User software must have the correct startup address. You don't need to change a code, just adjust ORIGIN and LENGTH of the ROM section in the linker script.

closed 2020-11-02 by curiousmindsdev