dmitrystu/sboot_stm32 #26: User firmware stuck or not starting after reset.

curiousmindsdev, 2020-11-03:
I have a demo firmware toggling 6 LEDs in sequence. I uploaded it using dfu-util but it does not start. First LED turns on and it stops there. I'm guessing that I forgot to add something in my code. DFU_APP_START is set to 0x08001000 and DFU_APP_SIZE is set to 126976 in config file. Also I set  `FLASH    (rx)    : ORIGIN = 0x08001000,   LENGTH = 124K` in my demo project. I've uploaded my demo project along with `bootloader.map` and `config.h` file to Google Drive. Can you check if something is wrong? 

https://drive.google.com/drive/folders/17VR4t5-j2v8Tgzqlx1nVWkIIFzimskdh?usp=sharing

dmitrystu, 2020-11-03:
Tried to debug with your code. I was surprised that F0 series doesn't support ISR table relocation. I see no way to fix this.

dmitrystu, 2020-11-03:
Probably I have found a solution. But it needs to be tested.

dmitrystu, 2020-11-03:
Could you test with [this](https://github.com/dmitrystu/sboot_stm32/tree/fix/F0_ISR) branch?
In a few words, since if we can't set a VTOR we must handle ISR in the bootloader then do a jump to the application ISR.

curiousmindsdev, 2020-11-03:
IT WORKED!!! Thank you so much!!!

closed 2020-11-03 by curiousmindsdev

dzarda, 2020-11-03:
Nice solution. I think we want to have this merged