dmitrystu/sboot_stm32 #34: User code in bootloader

dzarda, 2021-06-02:
What is the "correct" procedure for injecting user code into the bootloader? For example I want to display some content on the display while the device is in DFU.

One solution might be adding user_init and user_poll to the existing main:

```C
dfu_init();
user_init(); ///
while(1) {
    usbd_poll(&dfu);
    user_poll(); ///
}
```

Another solution might be to allow disabling the default main (and then calling dfu/usbd from user source:

```C
#ifndef DFU_CUSTOM_MAIN
dfu_init();
while(1) {
    usbd_poll(&dfu);
}
#endif
```

dmitrystu, 2021-06-02:
It's up to you. It's also possible to add two weak empty functions like ```void user_init(void)``` before the main loop and ```void user_task(state or somewhat_to_track_progress)``` inside the main loop to generalize this feature.

dmitrystu, 2021-06-02:
I think it's a good idea to add ```void user_exit(void)``` before a system reset.

dzarda, 2021-06-03:
Yeah the weak definitions are probably the way to go.

dzarda, 2021-06-03:
Do we want user_init before OR after dfu_init?

dmitrystu, 2021-06-04:
I think it doesn't matter.