dmitrystu/sboot_stm32 #46: Compilation with -Werror

salyzyn, 2022-03-30:
I added -Werror, and had DFU_CAN_UPLOAD = _DISABLE:

src/bootloader.c:109:21: error: 'dfu_upload' defined but not used [-Werror=unused-function]
  109 | static usbd_respond dfu_upload(usbd_device *dev, size_t blksize) {
      |                     ^~~~~~~~~~

All this requires is a '#if (DFU_CAN_UPLOAD == _ENABLE) / #endif' around dfu_upload function.

Are you willing to investigate, and support -Werror?

dmitrystu, 2022-04-09:
Are there more issues with -Werror option ?

salyzyn, 2022-04-15:
One of the assumptions I made was that you had a matrix build that would catch all of them, and that you may be ignoring them (I know, rude of me to suggest that). I did not have a configuration matrix build to find them. This was the only one that hit me for my specific configuration (above: I turned off DFU_CAN_UPLOAD as a security measure).